uuvsimulator/uuv_simulator #364: Is there hydrodynamics in the models?

nm46nm, 2019-05-27:
Is there hydrodynamics in the models? Or it is just a model of water and there's not hydrodynamics?
If there is hydrodynamics, where can I find corresponding codes?

Sollimann, 2019-05-28:
Rigid-Body kinetics are already included in Gazebo:
Rigid-body kinetics are: M_RB * v_dot + C_RB * v = tau_RB

Hydrodynamics and hydrodynamic damping are added through the uuv_simulator:
-M_A * vr_dot - C_A * vr - D * vr = tau_hydro

Hydrostatics g(n) are included through free-floating plugin

amamory, 2019-05-28:
for hydrodynamics for USV (winds and water current) you might take a look a this project
https://github.com/disaster-robotics-proalertas/usv_sim_lsa
We have plans to support UAVs with winds 'soon'.
It does not support UUV yet and there is no near future plan to include it.

musamarcusso, 2019-05-30:
The underwater object plugin might be what you are looking for: https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_gazebo_plugins/uuv_gazebo_ros_plugins/src/UnderwaterObjectROSPlugin.cc

nm46nm, 2019-05-30:
> The underwater object plugin might be what you are looking for: https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_gazebo_plugins/uuv_gazebo_ros_plugins/src/UnderwaterObjectROSPlugin.cc

I found uuv_simulator/uuv_gazebo_plugins/uuv_gazebo_plugins/src/HydrodynamicModel.cc and I think it's about hydrodynamics. Thanks for your reply and I will have a look of UnderwaterObjectROSPlugin.cc as well.

closed 2019-05-30 by nm46nm