uuvsimulator/uuv_simulator #367: how to get the velocity of rexrov?

nm46nm, 2019-06-21:
I launch the start_pid_demo.launch, and control the rov by publishing velocities to the topic rexrov/cmd_vel. But how can I get the real velocities of the robot. I only found the ground_truth_to_tf/pose but there's no ground_truth_to_tf/velocity.

HashirZahir, 2020-05-23:
Hi, you can use the pose_3d plugin macro, which publishes a nav_msgs::Odometry. As a sample, when moving forward in the simulator, I get this on the /auv/odom_ned topic:

[image]

You can see that the linear speed is at 0.18m/s.

Kupofty, 2021-04-11:
@HashirZahir Hi, i'm quite new to ROS, how do you get the pose_3d plugin ? Do you need to install it ? 
Like nm46nm i can only see /rexrov/ground_truth_to_tf/pose in the rostopic list. Could you explain what I have to do ? 
Thanks =)