uuvsimulator/uuv_simulator #373: Error, trying launch start_demo_nmb_sm.launch in both lauv and eca_a9.

liqsuq, 2019-08-28:
I will try to launch start_demo_nmb_sm.launch after I succeed start_demo_auv_control.launch and start_demo_teleop.launch.

I launch world fine, but I got following error  when I launched start_demo_nmb_sm.launch.

Maybe the main cause is start_control_allocator.launch and invalid "output_dir" arg...
I couldn't know how I should modify. 

~~~
$ roslaunch eca_a9_gazebo start_demo_nmb_sm.launch 
... logging to /home/redhawk/.ros/log/251a3ff2-c966-11e9-8c40-14feb5ee288b/roslaunch-t5500-16592.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
redefining global property: gamma
when processing file: /opt/ros/melodic/share/eca_a9_description/urdf/eca_a9_sensors.xacro
included from: /opt/ros/melodic/share/eca_a9_description/robots/eca_a9_default.urdf.xacro
started roslaunch server http://t5500:36429/

SUMMARY
========

PARAMETERS
 * /eca_a9/control_allocator/base_link: base_link
 * /eca_a9/control_allocator/fin_config/fin_area: 0.04155
 * /eca_a9/control_allocator/fin_config/fluid_density: 1028.0
 * /eca_a9/control_allocator/fin_config/frame_base: fin
 * /eca_a9/control_allocator/fin_config/lift_coefficient: 3.0
 * /eca_a9/control_allocator/fin_config/lower_limit: -1.4
 * /eca_a9/control_allocator/fin_config/topic_prefix: fins
 * /eca_a9/control_allocator/fin_config/topic_suffix: input
 * /eca_a9/control_allocator/fin_config/upper_limit: 1.4
 * /eca_a9/control_allocator/output_dir: /opt/ros/melodic/...
 * /eca_a9/control_allocator/thruster_config/conversion_fcn: proportional
 * /eca_a9/control_allocator/thruster_config/conversion_fcn_params/gain: 4.9e-05
 * /eca_a9/control_allocator/thruster_config/conversion_fcn_params/input: [0, 1, 2, 3]
 * /eca_a9/control_allocator/thruster_config/conversion_fcn_params/output: [0, 1, 2, 3]
 * /eca_a9/control_allocator/thruster_config/frame_base: thruster_
 * /eca_a9/control_allocator/thruster_config/max_thrust: 700
 * /eca_a9/control_allocator/thruster_config/topic_prefix: thrusters
 * /eca_a9/control_allocator/thruster_config/topic_suffix: input
 * /eca_a9/control_allocator/timeout: -1
 * /eca_a9/control_allocator/update_rate: 10
 * /eca_a9/ground_truth_to_tf_eca_a9/child_frame_id: /eca_a9/base_link
 * /eca_a9/ground_truth_to_tf_eca_a9/footprint_frame_id: /eca_a9/base_foot...
 * /eca_a9/ground_truth_to_tf_eca_a9/frame_id: world
 * /eca_a9/ground_truth_to_tf_eca_a9/odometry_topic: /eca_a9/pose_gt
 * /eca_a9/ground_truth_to_tf_eca_a9/stabilized_frame_id: /eca_a9/base_stab...
 * /eca_a9/robot_description: <?xml version="1....
 * /eca_a9/robot_state_publisher/robot_description: /eca_a9/robot_des...
 * /eca_a9/rov_nmb_sm_controller/Kd: [24.3289994058185...
 * /eca_a9/rov_nmb_sm_controller/Ki: [0.00102327681525...
 * /eca_a9/rov_nmb_sm_controller/dubins/max_pitch: 0.09
 * /eca_a9/rov_nmb_sm_controller/dubins/radius: 25
 * /eca_a9/rov_nmb_sm_controller/inertial_frame_id: world
 * /eca_a9/rov_nmb_sm_controller/look_ahead_delay: 2.0
 * /eca_a9/rov_nmb_sm_controller/max_forward_speed: 3.0
 * /eca_a9/rov_nmb_sm_controller/min_thrust: 40
 * /eca_a9/rov_nmb_sm_controller/saturation: 200
 * /eca_a9/rov_nmb_sm_controller/slope: [0.99038586689920...
 * /eca_a9/rov_nmb_sm_controller/thrusters_only: False
 * /eca_a9/rov_nmb_sm_controller/use_stamped_poses_only: False
 * /eca_a9/send_waypoint_file/filename: /opt/ros/melodic/...
 * /eca_a9/send_waypoint_file/interpolator: dubins
 * /eca_a9/send_waypoint_file/start_time: -1
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
[... 143 lines cut ...]
ECA_A9 -- 2019-08-28 16:33:56,426 | INFO | wp_trajectory_generator | Setting new starting time, t=17.63 s
ECA_A9 -- 2019-08-28 16:33:56,426 | ERROR | dp_controller_local_planner | Error generating trajectory message
ECA_A9 -- 2019-08-28 16:33:56,426 | INFO | dp_controller_local_planner | STATION KEEPING MODE = OFF
ECA_A9 -- 2019-08-28 16:33:56,426 | INFO | dp_controller_local_planner | AUTOMATIC MODE = ON
ECA_A9 -- 2019-08-28 16:33:56,426 | INFO | dp_controller_local_planner | TRAJECTORY TRACKING = ON
ECA_A9 -- 2019-08-28 16:33:56,473 | INFO | wp_trajectory_generator | Waypoint interpolator initialized! Start time: 17.63 s
ECA_A9 -- 2019-08-28 16:34:22,626 | INFO | dp_controller_local_planner | /eca_a9/ - Trajectory completed!
ECA_A9 -- 2019-08-28 16:34:22,626 | INFO | dp_controller_local_planner | STATION KEEPING MODE = ON
ECA_A9 -- 2019-08-28 16:34:22,626 | INFO | dp_controller_local_planner | AUTOMATIC MODE = OFF
ECA_A9 -- 2019-08-28 16:34:22,627 | INFO | dp_controller_local_planner | TRAJECTORY TRACKING = OFF
ECA_A9 -- 2019-08-28 16:34:27,674 | INFO | dp_controller_local_planner | AUV STATION KEEPING
ECA_A9 -- 2019-08-28 16:34:27,676 | INFO | dp_controller_local_planner | STATION KEEPING MODE = ON
ECA_A9 -- 2019-08-28 16:34:27,677 | INFO | wp_trajectory_generator | Interpolation method set: cubic
[INFO] [1566977667.677558, 48.880000]: Initial rotation vector (quat)=[  8.38085085e-17   1.85375118e-03   3.75953978e-16   9.99998282e-01]
[INFO] [1566977667.678610, 48.880000]: Initial rotation vector (rpy)=(1.6901174083154135e-16, 0.0037075044800792237, 7.522225537474225e-16)
[INFO] [1566977667.680376, 48.880000]: PathGenerator::Setting initial rotation as=[  8.38085085e-17   1.85375118e-03   3.75953978e-16   9.99998282e-01]
ECA_A9 -- 2019-08-28 16:34:27,681 | INFO | wp_trajectory_generator | Setting new starting time, t=48.88 s
ECA_A9 -- 2019-08-28 16:34:28,216 | INFO | dp_controller_local_planner | Updating the trajectory information
ECA_A9 -- 2019-08-28 16:34:28,216 | INFO | dp_controller_local_planner | STATION KEEPING MODE = OFF
ECA_A9 -- 2019-08-28 16:34:28,216 | INFO | dp_controller_local_planner | AUTOMATIC MODE = ON
ECA_A9 -- 2019-08-28 16:34:28,216 | INFO | dp_controller_local_planner | TRAJECTORY TRACKING = ON
ECA_A9 -- 2019-08-28 16:34:28,222 | INFO | wp_trajectory_generator | Waypoint interpolator initialized! Start time: 48.88 s
~~~

any help?

liqsuq, 2019-08-28:
The model is certainly spawn, but not be controlled, only floating to the surface.

wee-flame, 2020-04-17:
I also faced the situation. when I run roslaunch eca_a9_gazebo start_demo_nmb_sm.launch .The eca_a9 did not be controlled, only floating to the surface. Have you solved this problem?