uuvsimulator/uuv_simulator #376: ResourceNotFound: uuv_simulation_wrapper

ccs-ros, 2019-09-07:
hi,when i run the command "roslaunch rexrov2_gazebo start_demo_pid_controller.launch teleop_on:=true joy_id:=0",the following erroe occurred!

ResourceNotFound: uuv_simulation_wrapper
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/ccs/catkin_uuv/src
ROS path [2]=/home/ccs/cartographer_ws/catkin_ws/src
ROS path [3]=/home/ccs/catkin_robot/src
ROS path [4]=/home/ccs/catkin_bluerov/src
ROS path [5]=/opt/ros/kinetic/share

what should i do to solve this problem? thank you!

rooshm, 2019-10-10:
Same error with uuv_simulator installed from binaries for melodic 

Resource not found: uuv_simulation_wrapper
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/roosh/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

ccs-ros, 2019-10-11:
I have solved this problem.Thank you!

ccs-ros, 2019-10-11:
I have solved this problem.Thank you! @rooshm

ccs-ros, 2019-10-11:
I have not put the package "uuv_simulation_evaluation" in the workspace before.So，this error  "Resource not found: uuv_simulation_wrapper" occured. @rooshm

max0619, 2019-10-11:
Hey, I have the same error. How can I add uuv_simulation_evaluation to the workspace?

ccs-ros, 2019-10-13:
Just open terminal and enter your workspace,then "git clone https://github.com/uuvsimulator/uuv_simulation_evaluation.git".The error mentioned above will disappear.But when i run the catkin_make in the workspace,i met a new trouble：
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "uuv_simulation_evaluation" with any of the following names:

    uuv_simulation_evaluationConfig.cmake
    uuv_simulation_evaluation-config.cmake

  Add the installation prefix of "uuv_simulation_evaluation" to
  CMAKE_PREFIX_PATH or set "uuv_simulation_evaluation_DIR" to a directory
  containing one of the above files.  If "uuv_simulation_evaluation" provides
  a separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  uuv_simulation_evaluation/uuv_smac_utils/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ccs/catkin_uuv/build/CMakeFiles/CMakeOutput.log".
See also "/home/ccs/catkin_uuv/build/CMakeFiles/CMakeError.log".
Makefile:5120: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
@max0619

max0619, 2019-10-17:
Ok, I solved the problem, too. I added the uuv_simulation_evaluation to my workspace as described above but i used catkin build (comes with catkin tools) instead of catkin_make (you eventually have to run catkin clean before using catkin build).

rooshm, 2019-11-01:
In case someone can't run catkin_build because their workspace has other components that need to be built, the following pull request https://github.com/uuvsimulator/rexrov2/pull/32 should work on non lunar workspaces

kamilcetin, 2022-01-12:
When I run
roslaunch rexrov2_gazebo start_demo_pid_controller.launch teleop_on:=true joy_id:=0
I have the similar problem as you had before;

Resource not found: uuv_simulation_wrapper
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/robotic/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

To solve this problem, I am doing;
~/catkin_ws/src$ git clone https://github.com/uuvsimulator/uuv_simulation_evaluation
then I am doing;
~/catkin_ws$ catkin_make
But I encounter this error;

CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "uuv_simulation_evaluation" with any of the following names:
    uuv_simulation_evaluationConfig.cmake
    uuv_simulation_evaluation-config.cmake
  Add the installation prefix of "uuv_simulation_evaluation" to
  CMAKE_PREFIX_PATH or set "uuv_simulation_evaluation_DIR" to a directory
  containing one of the above files.  If "uuv_simulation_evaluation" provides
  a separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  uuv_simulation_evaluation/uuv_smac_utils/CMakeLists.txt:4 (find_package)

What should I do now?

tiko5000, 2022-04-04:
You can try out the fix from this pull request:

https://github.com/uuvsimulator/uuv_simulation_evaluation/pull/54/files