uuvsimulator/uuv_simulator #383: Joint Control Widget

TheGreatLebowski, 2019-10-29:
Hello,

I started a project with uuvsimulator, yesterday I've built a simple project with a cube as body, and one propeller.

When I load the cube into Gazebo, and I click on PLUGINS-

> > <name>_<id>_thruster_model

 or 

> uuv_joint_state_publisher

, I've got this error : 

[Err] [JointControlWidget.cc:400] Error advertising topic 

> [/data://world/empty_underwater_model/name/plugin/name_id_thruster_model/joint_cmd]

Also, (I don't know if there is a link), when I select the model, and change the value for the joint (In Newton-meter), the whole object is turning, not just the propeller ?

An idea ? Thank you !

pxalcantara, 2020-05-17:
@TheGreatLebowski Are you still with this problem?

Mateus224, 2020-09-01:
hej I m getting this error too:
[Err]  [JointControlWidget.cc:400] Error advertising topic [/data://world/default/model/box/plugin/model_push/joint_cmd]

and i cant move the box (with roslauch it works but not over gzserver/ gzclient)
I get this problem which i try to solve since 1 day. Maybe somebody can help me that is the tutorial (very short):

http://gazebosim.org/tutorials?tut=plugins_model&cat=write_plugin

ccwss-maker, 2022-01-24:
Hi,I m getting this error too, but I solved the problem.
After I 'make' my program ,In the build folder, I can see a file named as 'libvelodyne_plugin.so' .
However when I run this program,I get the error "Failed to load plugin libvelodyne_plugin.so: libvelodyne_plugin.so: cannot open shared object file: No such file or directory".
Than I copy that file to /usr/lib/aarch64-linux-gnu/ .
Than, that error disappeared.But another error appeared "Topic [/data://world/default/model/my_velodyne/plugin/velodyne_control/joint_cmd] is not valid.
[Err] [JointControlWidget.cc:400] Error advertising topic [/data://world/default/model/my_velodyne/plugin/velodyne_control/joint_cmd]"
You can solve the error just by this two steps :
1.Open the file "xxx.world" you just set up
2.put <plugin name="velodyne_control" filename="./libvelodyne_plugin.so"/> instead of <plugin name="velodyne_control" filename="libvelodyne_plugin.so"/>.

You may not copy the file named as 'libvelodyne_plugin.so' to /usr/lib/aarch64-linux-gnu/