uuvsimulator/uuv_simulator #384: Exemple for controlling the vehicle

TheGreatLebowski, 2019-10-29:
Hello,

In the section _Configuring the thruster manager_ of the tutorial, it is written at the end 

> To command the vehicle, you have to start the thruster manager and publish a ROS Wrench message to the input topic /<ROBOT_NAME or ROBOT_NAMESPACE>/thruster_manager/input with the wrench command to the vehicle.

Can you add an example ? And also I don't understand where I should put the command ...

Thank you for your help !

TheGreatLebowski, 2019-10-30:
I've maybe found an example but he is not working :

```
rostopic pub /submarine/thrusters/0/thrust_wrench geometry/WrenchStamped "header:
seq: 0
stamp:
  secs: 0
  nsecs: 0
frame_id=''
wrench:
  force:
    x: value
    y: value
    z: value
  torque:
    x: value
    y: value
    z: value"
```

But this error appears : 

> /opt/melodic/lib/python2.7/dist-packages/rostopic/__init__.py:1782: TAMLLoad Warning: calling yaml.load() without Loader=... is deprecated, as the default to Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
>  pub_args.append(yaml.load(arg))
> publishing and latching message. 

https://github.com/yaml/pyyaml/wiki/PyYAML-yaml.load(input)-Deprecation

A solution is available on this link, but doesn't work on my machine ...

Is my example wrong ? Or there is a bug ?

EmmaShou, 2019-11-18:
I am also confused by the end of this tutorial, have you made any progress? Thanks!

TheGreatLebowski, 2019-11-18:
Hello @EmmaShou, I've find a solution, you can check my project https://github.com/TheGreatLebowski/submarine, I've write a simple documentation (Still in progress) where I've put some of the problem I got ! If you don't find what is your problem send me a message, maybe I can help you !

EmmaShou, 2019-11-19:
Hi Alexandre,

Thank you -- that's really helpful! Can I ask you another question? Have
you been able to make your submarine either follow way-points or move
according to a function? There are UUV simulator tutorials about this but
I'm having a difficult time getting those tutorials to work.

Thanks a ton,
Emma

TheGreatLebowski, 2019-11-20:
Hi Emma ! No I am currently working on it, when I will find a solution I will add it in my github project, but if you find something please let me know !

EmmaShou, 2019-11-20:
Thanks Alexandre!

If I find a solution I will let you know. Please let me know if you make
progress

pxalcantara, 2020-05-17:
@TheGreatLebowski Nice, could you, please create a PR with your new documentation? It would be very helpful to have your colaboration!

Thanks!