uuvsimulator/uuv_simulator #385: Error, when try to creat a new vehicle

zt-luo, 2019-11-23:
when i run the command below, 
`rosrun uuv_assistants create_new_robot_model --robot_name arms_no1`
i got this error:

```
Create new catkin package for a UUV robot description
	Robot name = arms_no1
	Catkin package name = arms_no1_description
Creating the catkin package...
Created file arms_no1_description/CMakeLists.txt
Created file arms_no1_description/package.xml
Successfully created files in /home/ztluo/uuv_simulator/arms_no1_description. Please adjust the values in package.xml.
Done!
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/uuv_assistants/create_new_robot_model", line 59, in <module>
    for d in os.listdir(template_path):
OSError: [Errno 2] No such file or directory: '/opt/ros/melodic/share/uuv_assistants/templates/robot_model'

```

dnakath, 2019-12-06:
The current uuv_assistant package seems to be broken in ROS-Melodic, resulting in

    Traceback (most recent call last):
      File "/opt/ros/melodic/lib/uuv_assistants/create_new_robot_model", line 59, in <module>
        for d in os.listdir(template_path):
    OSError: [Errno 2] No such file or directory: '/opt/ros/melodic/share/uuv_assistants/templates/robot_model'

To quick-fix, clone uuv-repo and copy the respective templates

    git clone https://github.com/uuvsimulator/uuv_simulator.git
    cd uuv_simulator/uuv_assistants
    sudo cp -r ./templates/ /opt/ros/melodic/share/uuv_assistants/

Of course, it would be nicer to update the packages or work on the source from the GIT instead of messing around with the packages.

zt-luo, 2019-12-06:
> The current uuv_assistant package seems to be broken in ROS-Melodic, resulting in
>
> ```
> Traceback (most recent call last):
>   File "/opt/ros/melodic/lib/uuv_assistants/create_new_robot_model", line [...]

thanks

dnageo, 2019-12-06:
It looks like there is already the pull request #377 to fix this issue.