uuvsimulator/uuv_simulator #386: Path and trajectory generators: rexrov2 doesn't move

EmmaShou, 2019-11-28:
When following along with the ["path and trajectory generator tutorial"](https://uuvsimulator.github.io/packages/uuv_simulator/docs/features/path_and_trajectory_generation/) , ****I can't get rexrov2 to move!****  

 I get the following error when I run "roslaunch uuv_control_utils start_helical_trajectory.launch  uuv_name:=rexrov2 n_turns:=2"

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Traceback (most recent call last):
  File "/home/catkin_ws/src/uuv_simulator/uuv_control/uuv_control_utils/scripts/start_helical_trajectory.py", line 67, in <module>
    raise rospy.ROSException('Service not available! Closing node...')
rospy.exceptions.ROSException: Service not available! Closing node...
[rexrov2/start_helical_trajectory-1] process has died [pid 19617, exit code 1, cmd /home/catkin_ws/src/uuv_simulator/uuv_control/uuv_control_utils/scripts/start_helical_trajectory.py __name:=start_helical_trajectory __log:=/home/.ros/log/9aaddb66-1204-11ea-974f-0800279d5f12/rexrov2-start_helical_trajectory-1.log].
log file: /home/.ros/log/9aaddb66-1204-11ea-974f-0800279d5f12/rexrov2-start_helical_trajectory-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Is anyone else having problems with the [path and trajectory tutorial](https://uuvsimulator.github.io/packages/uuv_simulator/docs/features/path_and_trajectory_generation/)?

Any ideas how I might fix this?

arturmiller, 2019-12-17:
With the current version of uuv_simulator it is working on my machine. Nevertheless, I was able to reproduce the error, and I think I know what you are missing. You have to run "roslaunch rexrov2_gazebo start_demo_pid_controller.launch" before "roslaunch uuv_control_utils start_helical_trajectory.launch uuv_name:=rexrov2 n_turns:=2". Maybe this could be made a little bit clearer in the tutorial.

Yifangx1, 2020-01-13:
Hello @arturmiller ,

I'm encountering the same problem for trajectory generators. I ran the commands as follows: 
1. "roslaunch uuv_gazebo_worlds empty_underwater_world.launch"
2. "roslaunch rexrov2_description upload_rexrov2.launch"
3. "roslaunch rexrov2_control start_sf_controller.launch"
4. "roslaunch rexrov2_gazebo start_demo_pid_controller.launch"
5. "roslaunch uuv_control_utils start_helical_trajectory.launch uuv_name:=rexrov2 n_turns:=2"

After I ran step 4, "ResourceNotFound: uuv_simulation_wrapper" was shown and after step 5, same issue occured as @EmmaShou mentioned above.

Any help is appreciated. Thank you in advance.

pxalcantara, 2020-05-17:
@Yifangx1 are you still having this problem?