uuvsimulator/uuv_simulator #387: Call for maintainers

musamarcusso, 2019-12-05:
Hello everyone,

as you might have noticed, I haven't had time to take care of the this project as before. To give some perspective, this project was born in the scope of the EU-Project SWARMs and from my research at the time. The project is over since 2017 and I have also been working with other projects since then. I really apologize for the lack of reaction of my part, I have been struggling to find the time since I don't work on the UUV simulator full time anymore. 

After a lot of discussions and deliberations on my side, now I can invite people to co-maintain the UUV Simulator project with me and hopefully help the project to stay alive :)

Some of the main tasks for maintainers would include
* Responding to issues
* Organizing releases
* Keeping the code up-to-date
* Reviewing pull requests

It is a lot of work, so I would be very happy to see if anyone would be interested to work in this project. There would be some topics to discuss and some legal issues that I have to explain regarding contribution agreements, so I would kindly ask that anyone interested to send me an [e-mail](mailto:musa.marcusso@de.bosch.com) (maybe use the subject **UUV Simulator - Call for Maintainers** so I can filter them out easier) so I can organize a call to explain everything in more detail. 

Again, I apologize for the time it took to have this done, there were a couple of things to figure out before this was possible.
I also want to thank everyone that has contributed until now. The contributions via pull requests will, of course, still be possible, as always :).

If you have any questions, please feel free to write it here or contact me.

musamarcusso, 2019-12-06:
Just a small additional information for clarification: the idea is not to have a person doing all the work. The objective is to have a nice team to keep the project running and share the load of tasks to keep everything up and running :)

pxalcantara, 2019-12-07:
I can't see the e-mail that you use in the message, do you still using your gmail e-mail?

choeppler, 2019-12-09:
@pxalcantara, I'd use the e-mail address used in the commit messages' signed-off-by lines, see e.g. in commit 7577c874

pxalcantara, 2019-12-09:
@choeppler Thanks!!!

hakuturu583, 2019-12-10:
Hi.
I am masaya kataoka and I am a software developer of OUXT Polaris.
(http://robotx.osaka/about/)
We are developing ROS2 based UUV navigation system.
We will try uuv tasks in the future, so I am interested in UUV simulator.
I want to maintain this package and enables all ROS2 users to use this package.

hakuturu583, 2019-12-10:
my e-mail address is ms.kataoka@gmail.com

bsb808, 2020-06-09:
@musamarcusso I apologize that I just saw this.  We are starting a new project, extending UUV Simulator and other underwater Gazebo projects: [Project Dave v1.0.0 Announcement](https://discourse.ros.org/t/underwater-robotics-simulation-project-dave-initial-release/14561).

Are you still interested in volunteer maintainers?

pxalcantara, 2020-06-09:
@bsb808 As far as I know, yes, but she can say better. I' m became a maintainer recently. It would be awesome to converge with the efforts and hands to work!!

musamarcusso, 2020-06-10:
Hi @bsb808! Sure, the more the merrier. Could we set up a call this week to discuss this? Or could you send me a suggestion of times you are free for a call?

arturmiller, 2020-06-11:
Hello, I'm not interested in becoming a maintainer, but I could spend some time into this project the next months. I would like to help upgrading the build system and the noetic release.

pxalcantara, 2020-06-12:
@arturmiller It would be nice to have your help!