uuvsimulator/uuv_simulator #388: USV using uuv_packages

davihdsantos, 2019-12-05:
Hi,

Has anyone ever build a USV simulation using the uuv_simulator environment?

Im trying to add a sailboat model but im having trouble setting up the dynamics with gazebo breaking all the time, so i want some previous project use as template..

Thanks!

HashirZahir, 2020-05-16:
I would recommend the [VRX simulator](https://github.com/osrf/vrx/) for USV simulation. It is stable and has support for many sensors including GPS, IMU, 3D lidar, cameras and has good documentation as well.

pxalcantara, 2020-05-17:
@davihdsantos wich kind of problem you're having? could you give more details?

musamarcusso, 2020-05-23:
Hi @davihdsantos, I did include a simple implementation for buoyancy model for USVs, but the VRX Simulator is more appropriate for that kind of simulation.

closed 2020-12-21 by davihdsantos