uuvsimulator/uuv_simulator #390: Negative GPS coordinate

s1887252, 2020-03-26:
While modifying the UUV sensor plugins for a project, I have found this:

`// Copy the output of Gazebo's GPS sensor into a NavSatFix message
  this->gpsMessage.latitude = -this->gazeboGPSSensor->Latitude().Degree();
  this->gpsMessage.longitude = -this->gazeboGPSSensor->Longitude().Degree();`

what is the reason for making negative the latitude and longitude?

Thank you for your availability.

nicolaje, 2020-03-31:
I have the same remark.

MohitGupta007, 2020-05-12:
Is there a reason for the negative signs?

pxalcantara, 2020-05-17:
@musamarcusso Can you help with this answer ?

musamarcusso, 2020-05-23:
This was an issue raised by another user (I don't recall the issue number but I will look for it) where Gazebo's GPS sensor (for which the sensor here is a ROS wrapper) was delivering the readings with the axes flipped. I will check it again.