uuvsimulator/uuv_simulator #392: IMU Linear Accelerations do not work correctly

s1887252, 2020-04-02:
As stated at this link (in a different context but about the same topic)

[https://answers.ros.org/question/12616/interpretation-of-imu-data-from-the-gazebo_ros_imu-plugin/](url)

the UUV IMU implementation (as far as the Linear Accelerations are concerned) is not correct.

musamarcusso, 2020-05-23:
Hi @s1887252, the IMU is adapted from the Rotors Simulator and basically reads the linear acceleration from link https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_sensor_plugins/uuv_sensor_ros_plugins/src/IMUROSPlugin.cc#L183
But I recall having problem with the linear acceleration from Gazebo using the ODE engine with the `quick` algorithm. I can try changing the default algorithm to `world` to see if it behaves differently.

HashirZahir, 2020-05-24:
I am not sure if this is related, but gazebo recently updated its plugin to report IMU orientation from world frame. I was facing some issues in my other project due to its old behaviour in terms of body orientation vs world orientation, but this pull request claims to fix it.

https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1051