uuvsimulator/uuv_simulator #394: How to publish the value of the sliding surface in sliding mode controller

Xiaoran807, 2020-04-20:
I am working on a new project and hope to design a sliding mode controller for rexrov vehicle. I hope to see the values of different variables such as sliding surface function _s_b in rov_mb_sm_controller.py and restoring force _g in vehicle.py. I have tried to create my own topic and tried a few other things, unfortunately non of them worked. I am fairly new to robotic field.

musamarcusso, 2020-05-23:
Hi @Xiaoran807 , could you send the snippet on how you tried to publish the values?

Xiaoran807, 2020-05-24:
I have got a way to publish the topics. I have another question not related to publishing topics. Instead of constant current, do you have plan to develop models for modelling ocean current that rotate like vortex or even turbulence? Any idea or suggestion for materials to look at this problem will also be good.

saeed-NA, 2020-10-19:
Hi @ Xiaoran807 

Do you mind sharing the the code you tried to publish the parameters?

Xiaoran807, 2020-10-30:
> Hi @ Xiaoran807
> 
> Do you mind sharing the the code you tried to publish the parameters?

Hi, I have solved the problem.