uuvsimulator/uuv_simulator #396: HOW to add multi eca_a9 in gazebo

wee-flame, 2020-04-30:
I try to use this launch file to load two eca_a9 in the ocean_waves world and use geomertric tracking control method to control these two eca_a9 to follow the way_point path.But there is only one eca_a9 follow the path.  Another don't move. 
So I check the rviz. It shows that "eca_a9_2/base_link No transform from [eca_a9_2/base_link] to frame [world]".  Such problem can also be found in the tf_tree. There is only tf from world to eca_a9_1/base_footprint then eca_a9_1/base_stablized then eca_a9_1/base_link. But eca_a9_2/base_link has no father frame.
`
<launch>
    <arg name="record" default="false" doc="Set this flag to true to call record.launch"/>
    <arg name="bag_filename" default="recording.bag" doc="Name of the output ROS bag file in case record is set to true"/>
    <arg name="use_ned_frame" default="false" doc="If true, uses the NED (North-East-Down) frame conversion. If false, ENU (East-North-Up) will be used per default."/>
    <arg name="gui_on" default="true"/>

    <include file="$(find uuv_gazebo_worlds)/launch/ocean_waves.launch"/>

  <include file="$(find eca_a9_description)/launch/upload_eca_a9.launch">
    <arg name="x" value="0"/>
    <arg name="y" value="0"/>
    <arg name="z" value="-20"/>
    <arg name="yaw" value="0"/>
    <arg name="namespace" default="eca_a9_1"/>
  </include>

  <include file="$(find eca_a9_description)/launch/upload_eca_a9.launch">
    <arg name="x" value="0"/>
    <arg name="y" value="0"/>
    <arg name="z" value="-10"/>
    <arg name="yaw" value="0"/>
    <arg name="namespace" default="eca_a9_2"/>
  </include>

  <include file="$(find eca_a9_control)/launch/start_geometric_tracking_control1.launch">
        <arg name="use_ned_frame" value="$(arg use_ned_frame)"/>
        <arg name="uuv_name" default="eca_a9_1"/>
  </include>

  <include file="$(find eca_a9_control)/launch/start_geometric_tracking_control2.launch">
        <arg name="use_ned_frame" value="$(arg use_ned_frame)"/>
        <arg name="uuv_name" default="eca_a9_2"/>
  </include>

  <group if="$(arg gui_on)">
      <node name="rviz" pkg="rviz" type="rviz" output="screen" args="-d $(find eca_a9_control)/rviz/eca_a9_control_2.rviz"/>
  </group>

  <include file="$(find uuv_control_utils)/launch/send_waypoints_file.launch">
    <arg name="uuv_name" value="eca_a9_1"/>
    <arg name="filename" value="$(find eca_a9_gazebo)/config/waypoint_set.yaml"/>
    <arg name="interpolator" value="lipb"/>
  </include>

  <include file="$(find uuv_control_utils)/launch/send_waypoints_file.launch">
    <arg name="uuv_name" value="eca_a9_2"/>
    <arg name="filename" value="$(find eca_a9_gazebo)/config/waypoint_set2.yaml"/>
    <arg name="interpolator" value="lipb"/>
  </include>

  
  <include file="$(find eca_a9_gazebo)/launch/record_demo.launch">
    <arg name="record" value="$(arg record)"/>
    <arg name="use_ned_frame" value="$(arg use_ned_frame)"/>
    <arg name="bag_filename" value="$(arg bag_filename)"/>
  </include>
</launch>
`

[image]
[image]
[image]

trash:///frames.pdf
Can anybody help me? Please!

wee-flame, 2020-05-03:
I solved this problem

closed 2020-05-03 by wee-flame

HashirZahir, 2020-05-21:
Hi, can I check how you managed to solve this problem? I seem to be facing a similar issue of disconnected TF trees when running multiple vehicles.