uuvsimulator/uuv_simulator #397: Multi Robot TF tree broken

HashirZahir, 2020-05-24:
Hi, I am well aware that the UUV Simulator is able to simulate multiple robots (one of its strong points). However, I have been unable to get the TF trees of the multi robot system to remain connected with a single common parent TF frame `world`. 

When running `roslaunch uuv_gazebo_worlds ocean_waves.launch`, only `world` and `world_ned` are published.

[image]

Next, if I upload my custom `auv4` URDF, I get connected TF tree from `world` to `auv4/base_link`.

[image]

So far all good. Similarly, if I were to restart the simulator and run `roslaunch uuv_gazebo_worlds ocean_waves.launch` again, but this time uploading my `auv3` vehicle, I get connected TF tree from `world` to `auv3/base_link`.

[image]

So far all good. However, if I were to try to run `roslaunch uuv_gazebo_worlds ocean_waves.launch` and then upload both the urdfs simulataneously via separate upload.launch files (template follows upload_rexrov.launch), I get a disconnected TF tree, with the second vehicle launched usually disconnected from the TF tree. In this case, auv3 was uploaded first followed by auv4, only resulting in `world` connected to `auv3/base_link` but disconnected from `auv4/base_link`. 

[image]

Similar thing happens if I reverse the order of upload (auv4 first then auv3), then `world` is connected to `auv4/base_link` but not `auv3/base_link`. 

I am probably doing something wrong since #396 covers the same issue but somehow the Original Poster (OP) managed to resolve the issue by himself/herself. Could someone guide on how this issue can be resolved to allow a fully connected TF tree? I currently understand that the `message_to_tf.launch` file helps to convert nav_msgs::Odometry into to ROS TF transforms. Any help is appreciated. Thanks!

pxalcantara, 2020-05-27:
Hi @HashirZahir I did not have time yet but I'll try to reproduce your error and find a solution

Joel-Lindsay, 2020-06-29:
@HashirZahir I'm also interested in this same problem. Im attempting to launch 3 Eca_a9 vehicles simultaneously in the same manner after running _roslaunch uuv_gazebo_worlds ocean_waves.launch_. I can get at most two Eca_a9 vehicles working at once -- Have you had any further progress on this issue?

HashirZahir, 2020-07-02:
Could you share the steps to replicate to get the 2 Eca_a9 vehicles running together (and their respective TF trees?) In my case I am able to spawn both vehicles in the simulation but the TF tree is broken / disconnected, making it harder to perform multi-auv calculations/algorithms

HashirZahir, 2020-07-11:
Any luck @pxalcantara ?

pxalcantara, 2020-07-13:
Sorry @HashirZahir, but not yet, I've been very busy but I'll try to check this week.

Joel-Lindsay, 2020-07-15:
> Could you share the steps to replicate to get the 2 Eca_a9 vehicles running together (and their respective TF trees?) In my case I am able to spawn both vehicles in the simulation but the TF tree is broken / disconnected, making it harder to perform multi-auv calculations/algorithms

Sorry for the slow response. I have a launch file for launching 3 (or more) Eca UUVs under different namespaces using the upload.launch. I use another launch file to launch the geometric tracking controller for each of the 3 vehicles. Without knowing why, this generally works for 2/3 UUVs, with the 3rd being unresponsive and broken (as you have described)... So I can control, send waypoints, echo all topics for 2 UUVs. Note that this isn't always consistent and sometimes only one UUV will spawn correctly. I unfortunately don't have any more insight to the issue.

HashirZahir, 2020-09-05:
Has anyone managed to gain any insight on this issue? Multi-Robot capabilities of this simulator are amazing but I feel like some simple TF tree script is causing the broken TF tree, although I have not been able to find the source of the issue.

ErinWetter, 2020-10-02:
@HashirZahir 
I was having the same issue as @Joel-Lindsay with broken links with the eca_a9s. I fixed this by editing the file /eca_a9/eca_a9_description/robots/eca_a9_default.urdf.xacro. 

inertial_reference_frame="$(arg inertial_reference_frame)"
was changed to:
inertial_reference_frame="world"

This fixed my tf trees which also were disconnected.

HashirZahir, 2020-10-03:
Hi, @ErinWetter do you mind sharing a picture of the connected TF tree?

ErinWetter, 2020-10-03:
Here is the top of a tf tree with 4 working eca_a9: 

[image]

Going down to only two vehicles we can see the base link:

[image]

I haven't figured out why the frame links between broadcasters: ground_truth_to_uuv# and /uu#/robot_state_publisher were not connecting before hardcoding the robot state publisher to match but I'm not familiar with the xacro/urdf robot model creation process. 

I had run some tests on a single vehicle comparing the order of launching the broadcasting nodes on the Eca and RexRov (which has not had the broken tf tree issues and I've been using for multi-vehicle simulations). For a single Eca_a9, the tf tree would be severed at /base_link if I started the /ground_truth_to_uuv# node in terminal rather than within the "upload_eca_a9.launch" file. While doing the same with the RexRov /base_link connected with all launch conditions.