uuvsimulator/uuv_simulator #398: Noetic release

Timple, 2020-05-25:
Would you mind releasing these packages also for the newest distribution ROS Noetic? :)

musamarcusso, 2020-05-25:
I'm on it ;)

Timple, 2020-05-25:
Well that is very fast! :D

arturmiller, 2020-06-03:
Not all dependencies are released for noetic yet, e.g. joy and joy_teleop, needed by uuv_teleop.
https://github.com/ros-drivers/joystick_drivers/issues/170
https://github.com/ros-teleop/teleop_tools/issues/51

musamarcusso, 2020-06-05:
that is true, the first step would be to add a job on Travis so that we can check when the dependencies can be all found.

arturmiller, 2020-06-05:
I have done some experiments to update the Travis build. https://github.com/arturmiller/uuv_simulator/tree/noetic
https://travis-ci.org/github/arturmiller/uuv_simulator/jobs/693993787
It is not easy to implement that, because industrial_ci has made big changes. There are no BEFORE_SCRIPT and AFTER_SCRIPT arguments anymore and they have moved to colcon instead of catkin.
As I know catkin_tools is also not available for noetic, yet.

musamarcusso, 2020-06-05:
yes, I saw that they made a `legacy` version available with the old setup. I did some experiments locally on my machine here, there are a few things to check yet.

arturmiller, 2020-06-09:
I am now able to build and test uuv_simulator with the recent industrial_ci changes:
https://github.com/arturmiller/uuv_simulator/tree/noetic
https://travis-ci.org/github/arturmiller/uuv_simulator/builds/696565670
Most changes are straightforward but, there is one issue, which requires some form of refactoring. The tests of uuv_gazebo_ros_plugins require uuv_descriptions. This results in a circular dependency. For now I have removed the tests in uuv_gazebo_ros_plugins.
Apart from that I am not sure why the run_tests.sh script is used instead of the build in procedure from industrial_ci. I have used the build in test runner, because catkin_tools is not yet released in Ubuntu 20.04. It seems to work well.
If you want to merge this, I would clean up my commits and create multiple issues and PRs, where we can discuss everything if necessary.

arturmiller, 2020-06-09:
I forgot to mention. The packages joy and joy_teleop are still not available. For now, to make the travis builds succeed, I have commented these dependencies in the package.xml. Building and testing don't require them. But it is still blocking the noetic release.

purvikpatel, 2020-07-20:
@arturmiller Any updates? 
Can I use this repo for ROS noetic now since joy and joy_teleop are now available for ROS noetic ?

ethanjpark, 2020-07-21:
Bump. Would be very interested to know if UUV Sim is good to go (or would be good with some work on the user side when installing) on Noetic. My place is fully on Python3 and I'm currently dealing with the headache of coercing Melodic to work with Python3.

arturmiller, 2020-07-23:
Yes joy and joy_teleop are available now for noetic. catkin_tools is still not properly ported, but it can be easily installed from source with pip. 
@purvikpatel I would not recommend to use my fork directly. The build is passing, but I have not done any manual testing. I am sure it wont work, as the test coverage is not very high. In addition this branch is full of "exploratory" commits, because i was not able to exactly duplicate the travis build locally.
I have hoped for more interest from the maintainers on this, but they seem to be quite busy. My last issue #406 got nearly no attention.
I would like to see this repo being updated, but I could also create a working fork for noetic and Python3 (if the interest in general is large enough and if the maintenance of the repo keeps missing). However, I would not be able to maintain this project further and I cannot release any debian packages.

Timple, 2020-07-24:
> catkin_tools is still not properly ported, but it can be easily installed from source with pip.

Actually I think this is fixed. From ubuntu 20.04 I can simply do `sudo apt install python3-catkin-tools`

blaine141, 2020-09-01:
Any update on the progress of this? I tried the repo @arturmiller made and it is giving me a `free(): double free detected in tcache 2` error in gazebo.

Timple, 2020-11-09:
catkin tools is operational in Noetic now. 
Is someone willing to release this for Noetic? :slightly_smiling_face:

my-name-is-D, 2020-11-30:
you are doing an amazing job! I thanks this actual release: https://github.com/uuvsimulator/uuv_simulator/pull/423/commits/66474bd3648f9fa2d0cf1d42e153ee08f3d3da7e#diff-71b515d752a471114f16806cd6bf541d62e8d81e79f4fbce6325c45613561015 
for adapting the melodic version of uuv to noetic, can't wait for the official release. You are awesome.
 (posting this here for those trying desperately to make it work on noetic for the time being).

amaarquadri, 2021-01-24:
Hey everyone. Until there is an official release for noetic, does anyone have a working fork of the repo or a set of instructions that can be used to install it? Appreciate all the awesome work being done here!

my-name-is-D, 2021-01-24:
check the link of the comment above ;) It's was added there to make your life easier.

amaarquadri, 2021-01-26:
Thanks! I tried setting everything up based on that link and looking around the different pull requests and their comments. I got most of the way there but I'm getting the following error after building everything and trying to run it:

`ERROR: cannot launch node of type [message_to_tf/message_to_tf]: message_to_tf`

I compiled the process that I used here: https://github.com/Waterloo-Aquadrone/aquadrone2020_dev_workspace/blob/master/noetic_setup.md.

There's a good chance that I might just be making a dumb mistake. I would really appreciate any help at all. Thanks so much for all the awesome work!

my-name-is-D, 2021-01-26:
well, as far as i know, message_to_tf refers only to a launch and a c file in uuv_description ~to double check~. 
 If everything compiled you may assume you correctly installed  uuv_simulation. 

 I invite you to ask this question in ros answers (check:https://answers.ros.org/question/67488/roslaunch-error-error-cannot-launch-node-of-type/) or stackoverflow.

blaine141, 2021-02-20:
I am trying to implement this Neotic release and am running into a segfault when I add my robot. I have narrowed it down to whenever I add a sensor. I can spawn my robot without any sensors and it floats and works. The same code works in the melodic release but not in noetic. 

Here's an example of a block of code that will cause a segfault. I'm almost certain namespace and inertial_reference_frame exist as other code works.
``` 
<xacro:include filename="$(find uuv_sensor_ros_plugins)/urdf/sensor_snippets.xacro"/>

<xacro:default_imu_macro
    namespace="${namespace}"
    parent_link="${namespace}/base_link"
    inertial_reference_frame="${inertial_reference_frame}">
    <origin xyz="0 0 0" rpy="0 0 0"/>
</xacro:default_imu_macro>
```

Gazebo just prints "Segmentation fault (core dumped)" and then the process dies. Does anyone have any ideas?

blaine141, 2021-02-22:
If anyone else is having this problem, add `return true;` to line 144 of ROSBasePlugin.cc. 
I also had lots of errors related to ned. As a temporary solution, I deleted lines 207 and 208 from UnderwaterObjectROSPlugin.cc

hidmic, 2021-06-07:
Bumping this thread again :) @musamarcusso will you or any other maintainer release these packages for Noetic in the near term? If not, or if you're busy, let me know how may I help.

pxalcantara, 2021-06-07:
@hidmic, unfortunately, I don't have full access to the repository to manage the porting to Noetic, maybe waiting for some answer from @musamarcusso or working in some fork :/

hidmic, 2021-06-07:
Hi @pxalcantara! Long time no see :) 

> maybe waiting for some answer from @musamarcusso or working in some fork

I see. Not forking (i.e. not splitting efforts across copies of this repo) would be best IMHO.

Anishamohapatra6, 2021-11-10:
Unable to locate package ros-neotic-uuv-simulator
I am getting this error while I am trying to use the uuv_simulator which is available for melodic but I have ubuntu 20.04 and neotic I am getting this error.
It would be really great if this error is fixed as soon as possible .

jake3991, 2022-03-18:
Any update on a noetic release?

romulogcerqueira, 2022-04-14:
I am also interested in any updates of UUV to Noetic / Foxy

Ar-x-is, 2022-12-08:
Any update on an official noetic release?

Pranav-Malpure, 2022-12-14:
> Thanks! I tried setting everything up based on that link and looking around the different pull requests and their comments. I got most of the way there but I'm getting the following error after building everything and trying to run it:
> 
> `ERROR: cannot launch node of type [message_to_tf/message_to_tf]: message_to_tf`
> 
> Also (not sure if its related), but Gazebo is just showing a black screen, even though my code is actually running.
> 
> I compiled the process that I used here: https://github.com/Waterloo-Aquadrone/aquadrone2020_dev_workspace/blob/master/noetic_setup.md.
> 
> There's a good chance that I might just be making a dumb mistake. I would really appreciate any help at all. Thanks so much for all the awesome work!

Hey, thanks for the compilation, it really helped. But I was getting build error while doing 'catkin build' for the packages 'uuv_gazebo_ros_plugins' and 'uuv_sensor_ros_plugins' and as a result 7 other packages were getting abandoned. Any workaround for this? Anyone?

my-name-is-D, 2022-12-15:
@Pranav-Malpure , your first issues about blackscreen etc were listed as a bug of aquadrone. So as their package seems different from this one, there is a chance the issue lieswith the aqua package? So try asking there as well.

In any case, if you want anyone to solve your issue You need to add the bug message or something. Else there could be 1000 reasons why it doesn't work. :)

Pranav-Malpure, 2022-12-15:
Thanks for replying. Firstly, I skipped the step regarding the aqua drone setup since I just wanted UUV. 
Sorry for not adding exact error messages. This was my first post. I will keep that in mind from next time. Here is the snapshot of the terminal:
[image]
[image]

my-name-is-D, 2022-12-15:
I can be wrong, but that looks like a resources management issue. Looking your error on Internet its often linked to "compiler is getting killed by the OOM-killer due to using to many resources."
If it was a code error you would get a description of the error or at the very least the line it failed at.