uuvsimulator/uuv_simulator #412: Plans for migrating to ROS2

Michael-Equi, 2020-09-21:
Are there currently any plans or interest in migrating/porting uuv simulator to ROS2?

pxalcantara, 2020-09-21:
Hi @Michael-Equi, yes there are plans to migrate. We don't know yet when exactly but we have these plans.

LoicLiquid, 2020-09-30:
Hello @Michael-Equi 
With my colleague @jpliquid, we are developing [Plankton](https://github.com/Liquid-ai/Plankton), a simulator for maritime robotics researchers. 
As a first step on this project, we modified uuv_simulator to make it compatible with ROS 2 Eloquent (we will release soon a version compatible with Foxy). Feel free to look at it and test it if it answers your need, any feedback on our project would be very welcome.

Michael-Equi, 2020-09-30:
@LoicLiquid That looks like an awesome project. I'll definitely try it out.