uuvsimulator/uuv_simulator #418: How to enable angular motion of the uuv models in Gazebo?

Gowresh7, 2020-10-05:
I am simulating the LAUV model in gazebo with teleop through joystick. What I noticed was that the model never undergoes angular motion, i.e., It only either translates to the left or right instead of changing its heading towards the direction (Screenshot attached)
[image]

 How to enable the model with angular motions?

brschettini, 2020-10-11:
Hi,

Could you kindly share you urdf?

Gowresh7, 2020-10-12:
The following are my description files of LAUV model:-
1. base.xacro:-
`<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
  
  <xacro:include filename="$(find uuv_descriptions)/urdf/common.urdf.xacro"/>
  
  <xacro:include filename="$(find uuv_sensor_ros_plugins)/urdf/sensor_snippets.xacro"/>
  
  <xacro:include filename="$(find uuv_gazebo_ros_plugins)/urdf/snippets.xacro"/>
  
  <xacro:include filename="$(find lauv_description)/urdf/snippets.xacro"/>

  

  <xacro:property name="mass" value="19.6"/>
  
  <xacro:property name="cog" value="0 0 0"/>
  
  <xacro:property name="rho" value="1028"/>

  
  <xacro:property name="visual_mesh_file" value="file://$(find lauv_description)/meshes/body.dae"/>

  
  <xacro:property name="collision_mesh_file" value="file://$(find lauv_description)/meshes/body.stl"/>

  
  <xacro:macro name="lauv_base" params="namespace *gazebo inertial_reference_frame">

      
      <link name="${namespace}/base_link">
        
        <inertial>
          <mass value="${mass}" />
          <origin xyz="${cog}" rpy="0 0 0"/>
          <inertia ixx="${0.5 * mass * lauv_radius * lauv_radius}" ixy="0" ixz="0"
                   iyy="${0.083333333 * mass * 3 * lauv_radius * lauv_radius + 0.083333333 * mass * lauv_length * lauv_length}" iyz="0"
                   izz="${0.083333333 * mass * 3 * lauv_radius * lauv_radius + 0.083333333 * mass * lauv_length * lauv_length}" />
        </inertial>

        <visual>
          <origin xyz="0 0 0" rpy="0 0 0"/>
          <geometry>
            <mesh filename="${visual_mesh_file}" scale="1 1 1" />
          </geometry>
        </visual>

        <collision>
          <origin xyz="0 0 0" rpy="0 0 0"/>s
          <geometry>
            <mesh filename="${collision_mesh_file}" scale="1 1 1" />
          </geometry>
        </collision>
      </link>

    <gazebo reference="${namespace}/base_link">
      <selfCollide>false</selfCollide>
    </gazebo>

    <gazebo>
        <plugin name="talker" filename="libtalker.so">
            <name_link_to_change_mass>lauv/base_link</name_link_to_change_mass>
        </plugin>
    </gazebo>

    
    <xacro:insert_block name="gazebo"/>

    
[... 329 lines cut ...]
  <xacro:property name="fin_drag_coefficient" value="1.98"/>
  
  <xacro:macro name="fin_macro" params="namespace fin_id *origin mesh_filename">
      <xacro:fin_quadratic_macro
        namespace="${namespace}"
        parent_link="${namespace}/base_link"
        fin_id="${fin_id}"
        min_joint_limit="${-90.0 * d2r}"
        max_joint_limit="${90.0 * d2r}"
        mesh_filename="${mesh_filename}"
        fin_dynamics_time_constant="0.2"
        lift_constant="${0.5 * fin_rho * fin_area * fin_lift_coefficient}"
        drag_constant="${0.5 * fin_rho * fin_area * fin_drag_coefficient}"
        current_velocity_topic="current_velocity">
        <xacro:insert_block name="origin"/>
      </xacro:fin_quadratic_macro>
  </xacro:macro>

  

    

    
    
</robot>`

Gowresh7, 2020-10-13:
Hi @brschettini, I hope this is what you had asked for.

Gowresh7, 2020-10-15:
@brschettini Is there a way to bring in the angular motion functionality?

nitianlei984, 2021-05-07:
@Gowresh7 Hi,
did you solve this problem?