uuvsimulator/uuv_simulator #421: Simulation of USV

codymarquardt, 2020-10-20:
I am attempting to simulate a USV in addition to the eca_a9 AUV. 

I used the create_new_vehicle link provided and am able to create the USV in gazebo, but the gazebo simulation quickly hangs up and dies with the following message:

gzserver: /home/codymarquardt/catkin_ws/src/uuv_simulator/uuv_gazebo_plugins/uuv_gazebo_plugins/src/UnderwaterObjectPlugin.cc:210: virtual void gazebo::UnderwaterObjectPlugin::Update(const gazebo::common::UpdateInfo&): Assertion `(!std::isnan(linearAccel) && !std::isnan(angularAccel))&&("Linear or angular accelerations are invalid.")' failed.
Aborted (core dumped)

I have overcome a few error messages prior to this one, but I cannot seem to get past this one. Any help on this would be appreciated.

A working example of a USV simulation would be the most helpful, but I have not been able to find one so not getting my hopes up

vSebas, 2020-11-05:
Hi! Master branch of [this ](https://github.com/vanttec/vanttec_uv_sim)repo has simulation of an USV. It's still a work in progress as some things don't work properly yet (don't try to run the shell script), but you can at least see the configuration used.

albuckley, 2022-10-11:
I have the same issue with Plankton, a fork of uuv_sim.

It happens a lot when I activate the control system and path following.

I try and keep my thumb on the throttle when switching from manual to auto and try and point the vessel in the general direction of the first waypoint. It helps.

I suspect there is an issue with large discontinuities in control signals. Jumping from zero velocity or large rotations.

Looking at graphs I can see outrageously large velocity are produced just prior to the crash.