uuvsimulator/uuv_simulator #424: keyboard velocity operation

glakshay, 2020-11-23:
Hi, I am trying to use `key_board_velocity.launch` to control the vehicle but the this doesn't move the vehicle and the keys pressed do not immediately correspond to the terminal. I am using the following commands in order:

1. **`roslaunch uuv_gazebo_worlds empty_underwater_world.launch`**
2. **`roslaunch uuv_descriptions upload_rexrov.launch mode:=default x:=0 y:=0 z:=-20 namespace:=rexrov`**
3. I tried two variations for the keyboard operation but both are not working **`roslaunch uuv_control_cascaded_pid key_board_velocity.launch model_name:=rexrov`** or
**`roslaunch uuv_control_cascaded_pid key_board_velocity.launch uuv_name:=rexrov model_name:=rexrov`** 
Also, on pressing **_ctrl+c_**, I get the error:
_

> [rexrov/keyboard_uuv_velocity_teleop-4] process has died [pid 61583, exit code 255, cmd /home/lak/catkin_ws/src/uuv_simulator/uuv_teleop/scripts/vehicle_keyboard_teleop.py output:=/rexrov/cmd_vel __name:=keyboard_uuv_velocity_teleop __log:=/home/lak/.ros/log/3087991e-2dcb-11eb-82db-000c29a26e1a/rexrov-keyboard_uuv_velocity_teleop-4.log].
> log file: /home/lak/.ros/log/3087991e-2dcb-11eb-82db-000c29a26e1a/rexrov-keyboard_uuv_velocity_teleop-4*.log

_

Thanks for the help!

MakinoharaShouko, 2020-12-07:
I tried to reproduce the problem you encountered and it did not work. I found that there was an error complaining that the node with the type `vehicle_keyboard_teleop.py` could not be found. It seems like there was a line missing in the `/uuv_teleop/CMakeList.txt`. You can try to add a line `scripts/vehicle_keyboard_teleop.py` into the `catkin_install_python` part and do `catkin build` then see if it works.

glakshay, 2020-12-07:
> I tried to reproduce the problem you encountered and it did not work. I found that there was an error complaining that the node with the type `vehicle_keyboard_teleop.py` could not be found. It [...]

Thanks, i will try. Did it work for you then ?

MakinoharaShouko, 2020-12-07:
Yeah, I solved the problem on my computer by adding the line, but I am not sure if it will fix everything for you.