uuvsimulator/uuv_simulator #425: UUV is frozen at spawn

CraigWang1, 2020-11-26:
Hi,

I have run into this odd issue lately; sometimes, around 50% of the time, my robot will spawn in Gazebo, but it is stuck and unable to move. It also appears that for some reason the thrusters have not spawned in for some reason, and I am very puzzled.

Even when I publish thrust messages to thruster topics such as `/uuv/thrusters/0/input` it does not move the robot, whereas when it spawns normally, publishing to that topic does move the robot. 

[image]
[image]

Has anyone experienced any similar issues?

My apologies in advance if this is a simple question, but I have looked through various files and I have yet to find the problem.

Thank you in advance!

MakinoharaShouko, 2020-12-07:
Is there some way to reproduce this problem? It is hard to guess why just from the description.

CraigWang1, 2020-12-07:
@MakinoharaShouko Thank you very much for the response. I checked with a fellow team member and he said that it did not crash on his end, which leads me to believe that this occurs because my personal hardware sometimes cannot handle the Gazebo load. For reproducing the problem, I suspect this is the issue, so I think I can close this issue.

CraigWang1, 2020-12-08:
After searching online, it seems that this has fixed my issue:

In snippets.xacro, I changed
```
      <inertial>
        <mass value="0.001" />
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <inertia ixx="0.000000017" ixy="0.0" ixz="0.0"
               iyy="0.000000017" iyz="0.0"
               izz="0.000000017" />
      </inertial>
```
to
```
      <inertial>
        <mass value="0.001" />
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <inertia ixx="0.001" ixy="0.001" ixz="0.001"
               iyy="0.001" iyz="0.001"
               izz="0.001" />
      </inertial>
```
I also made sure that my inertia values for my robot in base.xacro were also at least 0.001, and for some reason it works!
Closing this issue.

closed 2020-12-08 by CraigWang1